NoobishSVK/fm-dx-webserver #1: Can't run it on Arch linux

zygmund2000, 2024-01-20:
Hi,

I'm trying run this code on Arch linux but libraries are not compatible, I have so many errors like that:

`fm-dx-webserver]$ npm install
npm WARN deprecated har-validator@5.1.5: this library is no longer supported
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated request@2.88.2: request has been deprecated, see https://github.com/request/request/issues/3142
npm ERR! code 1
npm ERR! path /home/tescik/aur/fm-dx-webserver/node_modules/ffi-napi
npm ERR! command failed
npm ERR! command sh -c node-gyp-build
npm ERR! make: Wejście do katalogu '/home/tescik/aur/fm-dx-webserver/node_modules/ffi-napi/build'
npm ERR!   CC(target) Release/obj.target/nothing/../node-addon-api/nothing.o
npm ERR! rm -f Release/obj.target/../node-addon-api/nothing.a Release/obj.target/../node-addon-api/nothing.a.ar-file-list; mkdir -p `dirname Release/obj.target/../node-addon-api/nothing.a`
npm ERR! ar crs Release/obj.target/../node-addon-api/nothing.a @Release/obj.target/../node-addon-api/nothing.a.ar-file-list
npm ERR!   COPY Release/nothing.a
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/prep_cif.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/types.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/raw_api.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/java_raw_api.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/closures.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/x86/ffiw64.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/x86/ffi64.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/x86/unix64.o
npm ERR!   CC(target) Release/obj.target/ffi/deps/libffi/src/x86/win64.o
npm ERR! rm -f Release/obj.target/deps/libffi/libffi.a Release/obj.target/deps/libffi/libffi.a.ar-file-list; mkdir -p `dirname Release/obj.target/deps/libffi/libffi.a`
npm ERR! ar crs Release/obj.target/deps/libffi/libffi.a @Release/obj.target/deps/libffi/libffi.a.ar-file-list
npm ERR!   COPY Release/libffi.a
npm ERR!   CXX(target) Release/obj.target/ffi_bindings/src/ffi.o
npm ERR! make: Opuszczenie katalogu '/home/tescik/aur/fm-dx-webserver/node_modules/ffi-napi/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@10.0.1
npm ERR! gyp info using node@21.6.0 | linux | x64
npm ERR! gyp info find Python using Python version 3.11.6 found at "/usr/bin/python3"
npm ERR! gyp http GET https://nodejs.org/download/release/v21.6.0/node-v21.6.0-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v21.6.0/node-v21.6.0-headers.tar.gz
npm ERR! gyp http GET https://nodejs.org/download/release/v21.6.0/SHASUMS256.txt
npm ERR! gyp http 200 https://nodejs.org/download/release/v21.6.0/SHASUMS256.txt
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args '/usr/lib/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args 'binding.gyp',
npm ERR! gyp info spawn args '-f',
npm ERR! gyp info spawn args 'make',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/home/tescik/aur/fm-dx-webserver/node_modules/ffi-napi/build/config.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/usr/lib/node_modules/node-gyp/addon.gypi',
npm ERR! gyp info spawn args '-I',
npm ERR! gyp info spawn args '/home/tescik/.cache/node-gyp/21.6.0/include/node/common.gypi',
npm ERR! gyp info spawn args '-Dlibrary=shared_library',
npm ERR! gyp info spawn args '-Dvisibility=default',
npm ERR! gyp info spawn args '-Dnode_root_dir=/home/tescik/.cache/node-gyp/21.6.0',
npm ERR! gyp info spawn args '-Dnode_gyp_dir=/usr/lib/node_modules/node-gyp',
npm ERR! gyp info spawn args '-Dnode_lib_file=/home/tescik/.cache/node-gyp/21.6.0/<(target_arch)/node.lib',
npm ERR! gyp info spawn args '-Dmodule_root_dir=/home/tescik/aur/fm-dx-webserver/node_modules/ffi-napi',
npm ERR! gyp info spawn args '-Dnode_engine=v8',
npm ERR! gyp info spawn args '--depth=.',
npm ERR! gyp info spawn args '--no-parallel',
npm ERR! gyp info spawn args '--generator-output',
npm ERR! gyp info spawn args 'build',
npm ERR! gyp info spawn args '-Goutput_dir=.'
npm ERR! gyp info spawn args ]
npm ERR! gyp info spawn make
npm ERR! gyp info spawn args [ 'BUILDTYPE=Release', '-C', 'build' ]
npm ERR! ../deps/libffi/src/java_raw_api.c: In function ‘ffi_java_raw_call’:
npm ERR! ../deps/libffi/src/java_raw_api.c:317:3: warning: ‘ffi_java_raw_to_ptrarray’ is deprecated [-Wdeprecated-declarations]
npm ERR!   317 |   ffi_java_raw_to_ptrarray (cif, raw, avalue);
npm ERR!       |   ^~~~~~~~~~~~~~~~~~~~~~~~
[... 66 lines cut ...]
npm ERR! /home/tescik/aur/fm-dx-webserver/node_modules/node-addon-api/napi-inl.h:2021:38:   required from ‘static Napi::Function Napi::Function::New(napi_env, Callable, const char*, void*) [with Callable = void (*)(const Napi::CallbackInfo&); napi_env = napi_env__*]’
npm ERR! ../src/ffi.cc:60:37:   required from here
npm ERR! /home/tescik/aur/fm-dx-webserver/node_modules/node-addon-api/napi-inl.h:63:47: error: invalid conversion from ‘napi_finalize’ {aka ‘void (*)(napi_env__*, void*, void*)’} to ‘node_api_nogc_finalize’ {aka ‘void (*)(const napi_env__*, void*, void*)’} [-fpermissive]
npm ERR!    63 |   status = napi_add_finalizer(env, obj, data, finalizer, hint, nullptr);
npm ERR!       |                                               ^~~~~~~~~
npm ERR!       |                                               |
npm ERR!       |                                               napi_finalize {aka void (*)(napi_env__*, void*, void*)}
npm ERR! /home/tescik/.cache/node-gyp/21.6.0/include/node/js_native_api.h:516:43: note:   initializing argument 4 of ‘napi_status napi_add_finalizer(napi_env, napi_value, void*, node_api_nogc_finalize, void*, napi_ref__**)’
npm ERR!   516 |                    node_api_nogc_finalize finalize_cb,
npm ERR!       |                    ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~
npm ERR! make: *** [ffi_bindings.target.mk:117: Release/obj.target/ffi_bindings/src/ffi.o] Błąd 1
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack at ChildProcess.<anonymous> (/usr/lib/node_modules/node-gyp/lib/build.js:209:23)
npm ERR! gyp ERR! System Linux 6.6.12-1-lts
npm ERR! gyp ERR! command "/usr/bin/node" "/usr/lib/node_modules/node-gyp/bin/node-gyp.js" "rebuild"
npm ERR! gyp ERR! cwd /home/tescik/aur/fm-dx-webserver/node_modules/ffi-napi
npm ERR! gyp ERR! node -v v21.6.0
npm ERR! gyp ERR! node-gyp -v v10.0.1
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in: /home/tescik/.npm/_logs/2024-01-20T19_12_33_662Z-debug-0.log
`

Maybe u can help me with this issue?

NoobishSVK, 2024-01-21:
Which version did you try to run? I have posted a new version yesterday that doesn't require node-pre-gyp.

zygmund2000, 2024-01-21:
I just did git clone today and next try npm install, so maybe install script need some update?

NoobishSVK, 2024-01-21:
Please try it now, I have updated the files, it seems like package.json changes didn't come through on the first try. 
Also just in case, make sure your version of node.js is **21.5.0** or older. 

You can use the `node --version` command to check that.

zygmund2000, 2024-01-21:
I see some changes now, but I don't know is that ok and what's next?

npm install

added 91 packages, and audited 92 packages in 5s

12 packages are looking for funding
  run `npm fund` for details

1 moderate severity vulnerability

To address all issues, run:
  npm audit fix

Run `npm audit` for details.

HyperDX, 2024-01-21:
Try running it with ``node .``

zygmund2000, 2024-01-22:
Okej, so how to connect it with TEF6686 tuner? I have only serial port and I can't see any settings of serial port

andimik, 2024-01-22:
Why don't you use wifi? It's very easy and will significantly improve your reception as over serial you will have much more noise.

zygmund2000, 2024-01-22:
Because wifi doesn't work on my esp32 and I don't know why, just failed after trying to enable :(

andimik, 2024-01-23:
So you mean, your TEF is broken or have you never set it up for your wifi? Please connect to the temporary wifi ESP32... with your mobile and let TEF search for the correct wifi. 

Because this menu entry will only an already configured wifi setting, which has been deactivated before.

NoobishSVK, 2024-01-23:
That's where xdrd comes in, you need to have xdrd running in order to run the webserver. The webserver connects as a client to an already existing XDR server (either xdrd or web-hosted ESP32 TEF in case of Sjef's firmware).

zygmund2000, 2024-01-23:
In my esp32 wifi not starting when I'm trying to enable it in menu
[image]
Anyway I'll try xdrd instead of wifi and back later with info.

andimik, 2024-01-23:
Have you ever configured it?

zygmund2000, 2024-01-23:
No because I can't enable it

andimik, 2024-01-23:
It's not an issue, please go to the configure menu option and follow the instructions.

If you **never** configured wifi but want to activate it, then you get that `FAILED!` error.

zygmund2000, 2024-01-23:
Configuration also not working, so I can't follow instructions :(

andimik, 2024-01-23:
Which firmware do you have? 2.00 is recommended.

zygmund2000, 2024-01-23:
2.00

zygmund2000, 2024-01-23:
Wifi is not important at the moment, because xdrd works fine.
Let's back to node . and put some logs below:

rdsparser.register_pi(rds, callbacks.pi);
          ^

TypeError: Unexpected External value, expected callback_pi *
    at Object.<anonymous> (/home/tescik/aur/fm-dx-webserver/datahandler.js:154:11)
    at Module._compile (node:internal/modules/cjs/loader:1378:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1437:10)
    at Module.load (node:internal/modules/cjs/loader:1212:32)
    at Module._load (node:internal/modules/cjs/loader:1028:12)
    at Module.require (node:internal/modules/cjs/loader:1237:19)
    at require (node:internal/modules/helpers:176:18)
    at Object.<anonymous> (/home/tescik/aur/fm-dx-webserver/index.js:12:21)
    at Module._compile (node:internal/modules/cjs/loader:1378:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1437:10)

Node.js v21.6.0

NoobishSVK, 2024-01-23:
Please, read the "Getting started" guide on the main page of the project. Specifically this part:
`Please install node.js version 21.5.0 or older to make this work. Version >=21.6.0 is currently not working correctly.`

zygmund2000, 2024-01-25:
Finally with older node I have full success :)

NoobishSVK, 2024-01-26:
Perfect. If you find any more bugs, feel free to either create an issue here or just message us on Discord.

closed 2024-01-26 by NoobishSVK

zygmund2000, 2024-01-27:
It's ok ,  but only support for mobile devices is not good, don't match to screen and it is not scalable, some texts are outside of screen.