NoobishSVK/fm-dx-webserver #3: Task list

NoobishSVK, 2024-01-22:
We can use this issue to discuss the current tasks. 

- [ ] Make AF list clickable 
- [ ] Add RF+/IF+ (iQ/EMS) 
- [ ] Add 3LAS support (Low-Latency audio stream)
- [ ] Pack the entire project into a single binary so node.js isn't needed
- [ ] Make screenshots possible with a button/keypress
- [ ] Add password auth
- [ ] Improve auto frequency rounding on FM
- [ ] Modify frequency steps for AM and OIRT
- [ ] Improve UI for mobile devices
- [ ] Make RT copyable 
- [ ] Refactor code 
- [ ] Reset RDS (on R press)
- [ ] Accessibility features (for visually impaired people)
- [ ] Custom scrollbar (for the AF bar) 
- [ ] Web setup
- [ ] Broadcast on all IPs by default  
- [ ] Add admin password for remote settings management

andimik, 2024-01-24:
- defined band margins (FM, OIRT) resp. switching between bands
- tuning steps (100 kHz, 200 kHz)
- US/Europe switch
- export (or make streamable) RDS (with timestamp for RDSSpy) on different page or port
- decrease UI size for desktops
- don't place elements absolute but relative

PE5PVB, 2024-01-27:
Use mousewheel to go freq up/down.

NoobishSVK, 2024-02-01:
I am closing this task list due to making a new one using Trello - this task list is public and members are always welcome to give me the feedback on our Discord server. I will be slowly adding features requested by users here on GitHub as well. 
https://trello.com/b/OAKo7n0Q/fm-dx-webserver

closed 2024-02-01 by NoobishSVK