NoobishSVK/fm-dx-webserver #5: EBU special characters

andimik, 2024-01-22:
redsea has no problem with the letter `č`

```
{"pi":"0x9202","group":"2A","radiotext":"Več kot radio","prog_type":"Varied","tp":true}
```

but the radiotext on the webserver produces a line break here:

[image]

And German Umlaut `Ö` is missing

```
Jetzt in �1: Anklang (Wiederholung von 10:05 Uhr)    
�1 Club: (01) 501 70 370 (Mo-Fr, 9-17 Uhr)                      
```

kkonradpl, 2024-01-23:
Fixed in d3e013f60405ff9b6b79a1c9255b1ea04fe63925

NoobishSVK, 2024-01-23:
@kkonradpl fixed the issue, please test

andimik, 2024-01-23:
yes, confirmed, this fixed it (German and Eastern European characters).

```
Ö1 Service: Tel. (01) 501 70 371 (Mo-Fr, 8-21 Uhr)              
momentan viel Geld flieβt.                                      
Nächste Sendung: Gerald Finley & Julius Drake                   
```
```
Več kot radio
```