NoobishSVK/fm-dx-webserver #8: Disappearing PI-Code

andimik, 2024-01-24:
The PI Code disappears (TEF connected to webserver without explicit xdrd)

See https://discord.com/channels/1053804249651359765/1170013490661048411/1199101291754963026

# Video

[Bildschirmaufzeichnung vom 22.01.2024, 22:19:19.webm](https://github.com/NoobishSVK/fm-dx-webserver/assets/24510556/077af51e-c609-403a-9eae-ed67a26eb4d7)

# Reason

> because the PI that comes from xdrd gets overwritten by the PI from the librdsparser library, which is empty

Replaced `datahandler.js` by the following patched debug version: 
[datahandler.zip](https://github.com/NoobishSVK/fm-dx-webserver/files/14044120/datahandler.zip)

> The PI code is not always 4 digits

```
[INFO] a540
[INFO] 
[INFO] 
[INFO] a540
[INFO] a5
[INFO] 
[INFO] 
```

NoobishSVK, 2024-01-26:
A fix is planned and should come in the next few days. Thank you for reporting.

kkonradpl, 2024-01-27:
Fixed in https://github.com/NoobishSVK/fm-dx-webserver/commit/5450ab671d49b180a07cc195808d335ede4c5252

andimik, 2024-01-27:
Will check it tomorrow.

andimik, 2024-01-28:
I cannot start it with 8eb63eb5c4f1de6bd10b96ea80eb4fcedf30db8d

```
$ node .
internal/modules/cjs/loader.js:818
  throw err;
  ^

Error: Cannot find module 'command-exists-promise'
Require stack:
- /home/andreas/apps/fm-dx-webserver/index.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:815:15)
    at Function.Module._load (internal/modules/cjs/loader.js:667:27)
    at Module.require (internal/modules/cjs/loader.js:887:19)
    at require (internal/modules/cjs/helpers.js:74:18)
    at Object.<anonymous> (/home/andreas/apps/fm-dx-webserver/index.js:12:23)
    at Module._compile (internal/modules/cjs/loader.js:999:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1027:10)
    at Module.load (internal/modules/cjs/loader.js:863:32)
    at Function.Module._load (internal/modules/cjs/loader.js:708:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:60:12) {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/home/andreas/apps/fm-dx-webserver/index.js' ]
}
```

andimik, 2024-01-28:
with help from discord I made it. PI code is now working and will not disappear.

```
git pull --recurse-submodules
sudo npm install -g node-pre-gyp
sudo npm install

```
[image]

closed 2024-01-28 by andimik