NoobishSVK/fm-dx-webserver #9: Disconnecting - Unhandled 'error' event in node.js

andimik, 2024-01-24:
This error refers to `/usr/share/nodejs/events/events.js`

```
andreas@E535:~/apps/fm-dx-webserver$ node .
[23:10] [INFO] Web server is running at http://0.0.0.0:8080.
[23:10] [INFO] Connection to xdrd established successfully.
[23:10] [INFO] Authentication with xdrd successful.
[23:10] [INFO] Web client connected (192.168.43.252) [1]
[23:13] [DEBUG] Received message from client: 94.8
events.js:291
      throw er; // Unhandled 'error' event
      ^

Error: read ECONNRESET
    at TCP.onStreamRead (internal/stream_base_commons.js:209:20)
Emitted 'error' event on Socket instance at:
    at emitErrorNT (internal/streams/destroy.js:92:8)
    at emitErrorAndCloseNT (internal/streams/destroy.js:60:3)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  errno: 'ECONNRESET',
  code: 'ECONNRESET',
  syscall: 'read'
}
```

I hope the information is enough.

NoobishSVK, 2024-01-25:
It's just a message issue (as in it has no impact on the way the app works), but I will add xdrd disconnect messages as well.

andimik, 2024-01-25:
No, the hardware seems to get a command to soft "reboot" (but with audio on). So it is not only a message.

NoobishSVK, 2024-01-25:
ECONNRESET happens when the webserver loses connection to xdrd (in your case the tuner directly), therefore it can't send any commands. I think this might be an issue with the firmware itself.

NoobishSVK, 2024-01-27:
I have tried to add error handling to these messages, please check.

NoobishSVK, 2024-02-15:
After talking to Sjef, this seems more like a firmware issue - the side effect has been fixed. 
Since there has been no reply since then, I'm closing the issue.

closed 2024-02-15 by NoobishSVK