NoobishSVK/fm-dx-webserver #13: 3LAS stream

zygmund2000, 2024-01-28:
Current implementation of sound works but very unstable and mono, I've modified to stereo but still it is unstable too much sound drops on chrome and firefox.

NoobishSVK, 2024-01-29:
What's your hardware? Seems to work fine my Android phone, Linux laptop and a Windows PC.

zygmund2000, 2024-01-30:
It doesn't matter witch hardware I use to test laptop or smartphone because sound drop are always, sometimes it makes crash all node and I need to restart all. My platform for serve is linux but I think you are using windows because parameter for ffmpeg -dshow is windows sound api, please try linux.

NoobishSVK, 2024-01-31:
There are predefined defaults coming for linux via alsa and we're currently fiddling with the buffer settings, I will notify you once the commit gets published.

andimik, 2024-01-31:
Yes, please, I am interested, too.

NoobishSVK, 2024-02-01:
There is a new script for Linux, WAV fallback has also been disabled for Windows. Is there any noticeable change?

NoobishSVK, 2024-02-07:
It seems like you have opened a ticket directly with the 3LAS developer - the issue should be solved during the weekend, let's hope for the best!

NoobishSVK, 2024-02-12:
Are there any improvements with the latest commit? I have merged some changes from 3LAS directly.

zygmund2000, 2024-03-08:
Yes now it is very good so u can mark it as solved.

closed 2024-03-10 by NoobishSVK