NoobishSVK/fm-dx-webserver #15: Failed to load shared library: wrong ELF class: ELFCLASS64

cristianpuddu, 2024-01-29:
I tested fm-dx-webserver on hardware ARM platform (CPU: Allwinner H3, Quad-core Cortex-A7, Nano Pi device). Unfortunately,  with  Debian 12  or Armbian, when I start the server with "node ."  I get the following error. It seems to be a problem with Koffi, possibly due to a difference between 32 or 64-bit libraries. Tnx

```console
/home/cristian/FM-DX-Webserver/node_modules/koffi/index.js:562
  let lib = load(...args);
            ^

Error: Failed to load shared library: wrong ELF class: ELFCLASS64
    at module.exports.load (/home/cristian/FM-DX-Webserver/node_modules/koffi/index.js:562:13)
    at Object.<anonymous> (/home/cristian/FM-DX-Webserver/datahandler.js:9:19)
    at Module._compile (node:internal/modules/cjs/loader:1356:14)
    at Module._extensions..js (node:internal/modules/cjs/loader:1414:10)
    at Module.load (node:internal/modules/cjs/loader:1197:32)
    at Module._load (node:internal/modules/cjs/loader:1013:12)
    at Module.require (node:internal/modules/cjs/loader:1225:19)
    at require (node:internal/modules/helpers:177:18)
    at Object.<anonymous> (/home/cristian/FM-DX-Webserver/index.js:14:21)
    at Module._compile (node:internal/modules/cjs/loader:1356:14)
   
```

NoobishSVK, 2024-01-29:
You need to compile librdsparser from source by yourself in this case. The original .so library is an amd64 one, not arm64. 
https://github.com/kkonradpl/librdsparser

cristianpuddu, 2024-01-30:
thanks @NoobishSVK , I compiled the library. Now works, but no audio!
I'm not sure what is  the device name in  configuration file on Linux.
const audioDeviceName = "hw:1,0"
could this be correct?

arecord -l return:
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

NoobishSVK, 2024-01-31:
Perfect. Linux pre-defined flags will come in a commit today, so you won't have to set them yourself manually anymore. 
Since the compilation issue is fixed, i'm closing the issue. Thanks for testing!

closed 2024-01-31 by NoobishSVK