NoobishSVK/fm-dx-webserver #38: Help needed - transmitter info dissapeared

zygmund2000, 2024-03-08:
I don't know why but transmitter info suddenly dissapeared from my server, I tried to clean everything and setup from scratch and still  empty :(
Console log doesn't show any errors or warnings and other functions works correctly.
How someone some idea to trace this issue?

kkonradpl, 2024-03-09:
There is an ongoing diversification of the data sources at maps.fmdx.pl. The Polish (POL) database now comes from RadioPolska instead of FMLIST.
I'm waiting for a dedicated API endpoint to include the inactive stations and RDS PS which is required for station identification in FM-DX Webserver. It should be fixed soon.

NoobishSVK, 2024-03-10:
Try to restart your webserver and check if there's any change.

zygmund2000, 2024-03-15:
Yes by this time it always shows info.
BTW @kkonradpl can I copy that database to my own server?

NoobishSVK, 2024-03-16:
It's always possible to contact Konrad outside of this topic. You can also find him on our Discord server. Since the issue has been solved, I am locking this topic. Thank you for the report!

closed 2024-03-16 by NoobishSVK