NoobishSVK/fm-dx-webserver #39: Audio Quality resets to 192k every time you save other settings in the admin panel.

Overland-DX, 2024-03-09:
> From Discord:
>Kurer501 — i går kl. 17:41
>Possible bug in webserver. I'm running v1.1.2a. If bitrate is set to 256, and you later go inn and change something else like text and >save, the bitrate changes to 192.

I'll show you here:
https://github.com/NoobishSVK/fm-dx-webserver/assets/102610897/3eaef655-9900-46d9-bf9f-f3f2c498eefa

NoobishSVK, 2024-03-09:
Thank you, I will look into it further.

NoobishSVK, 2024-03-10:
Please check now, should be fixed in the latest update.

Overland-DX, 2024-03-11:
Looks like it works fine in the new version, but it was @kurer501 who reported this bug.

NoobishSVK, 2024-03-16:
Multiple people have reported that this issue has been fixed, therefore I'm locking this topic. Thank you very much for the report!

closed 2024-03-16 by NoobishSVK