NoobishSVK/fm-dx-webserver #40: Can't get v1.1.3 it to work with fm-dx-tuner.

Overland-DX, 2024-03-11:
I can't get the new version (1.1.3) to work with fm-dx-tuner (arduino nano+tef6686 module.
It connects to the correct comport, but it doesn't look like fm-dx web server and fm-dx tuner are talking at all.
I have tested with esp32-spi (fw2.00.6) gray china model and everything seemed fine there.

NoobishSVK, 2024-03-16:
I made some changes in 1.1.4, though I don't think they will help. I'll look into it further.

HME-Tef, 2024-03-18:
> I made some changes in 1.1.4, though I don't think they will help. I'll look into it further.

Can you make a update for the Aduino Nano TEF6686?
Or what code can I put in the source to make it work?
The Nano's does work with XDR-GTK so the communication is good.

Overland-DX, 2024-03-18:
I tested with v1.1.4 and it still doesn't work. No stress, just using xdrd for now.

DABtester, 2024-03-21:
Same here, tested with 1.1.4 and does not connect. Would be great if this could be resolved.

NoobishSVK, 2024-03-21:
I have added one more command and it seems to work properly on my ESP32+TEF using Konrad's firmware. The update will be out in a few minutes.

DABtester, 2024-03-22:
ARDUINO tested with 1.1.6, not yet working

Overland-DX, 2024-03-22:
I also tested with 1.1.6, but it still doesn't work.

NoobishSVK, 2024-03-28:
I have asked @kkonradpl for the initialization commands and I have tried to introduce them. Please let me know if there's any change. 
On top of that, you may need to try to change the baud rate in case 115200 doesn't work in server/index.js

Overland-DX, 2024-03-29:
I have tested 19200, 38400, 57600 but no response. I was at 19200 when I went back to xdrd, but then I had to go back to 115200 for xdrd to work again.

DABtester, 2024-04-01:
Tested with 1.1.9, but unfortunately still no connection.

- Looked at the baud rate settings of the COM port (9600) and set 9600 in index.js
- Manually changed the baudrate of the COM to 115200 and changed index,js to 115200

Just so that we are on the same page: When running the setup/wizard, the device that we have to choose is TEF6686/TEA685X with USB cable as Tuner connection... Correct?

NoobishSVK, 2024-04-05:
@Overland-DX  Thanks for testing, I will have to consult this with @kkonradpl further.
@DABtester I see, the device chosen shouldn't matter - it seems like there might still be something missing in the serial init script. I will try to find more ways to fix this.

NoobishSVK, 2024-07-16:
I made some changes, would it be possible for you to check now?

DABtester, 2024-07-16:
Tested, but cannot get it to work.

Using XDRD in the middle at the moment, so no hurry

Overland-DX, 2024-07-16:
I also tested. I do not get a response when contacting directly. Via xdrd it works fine.

NoobishSVK, 2024-07-18:
Sjef made a commit that fixed the issue for him. Can you try now?

DABtester, 2024-07-18:
Tested - unfortunately doesn't work for me

PE5PVB, 2024-07-18:
I'll try to build a Arduino version tonight to find the cause.

NoobishSVK, 2024-07-21:
I tried to disable RTS/DTR on the serial port, that may fix the issue.

NoobishSVK, 2024-08-02:
Are there any changes as of the current (1.2.6a) version for you? 
@DABtester @Overland-DX

Overland-DX, 2024-08-02:
I updated to v 1.2.6a but it's still the same. Still no response when direct contact is selected in settings.

Overland-DX, 2024-08-02:
The pull request #76  from AmateurAudioDude seems to work with nano+tef.

DABtester, 2024-08-02:
Same here, version from AmateurAudioDude works with nano

NoobishSVK, 2024-08-29:
Amazing, in that case we can close it as this was included in the webserver. Thank you very much for being so patient!

closed 2024-08-29 by NoobishSVK