NoobishSVK/fm-dx-webserver #41: Cannot tune with a password if the tuner is locked.

Overland-DX, 2024-03-11:
[image]

It may appear that I do not have the opportunity to tune when I am logged in with the tune password or admin password when the tuner is locked.
This is either if it is locked to admin or if you have not ticked the option "allow tuner control without password"

NoobishSVK, 2024-03-12:
I have checked in 1.1.3 and this seems to work. 

Tuner Lock enabled = I was only able to tune as an admin
Allow tuner control without password disabled = I was able to tune with either tune/admin password.

Overland-DX, 2024-03-12:
oki, i have locked me server for password only: The password is: changeme
Please take a look at my server.
http://overland.fmdx.no:45990/

NoobishSVK, 2024-03-19:
Seems like it's a serialport issue. I will look into this further.

NoobishSVK, 2024-03-21:
Please try now, seems to be fixed with 1.1.6.

NoobishSVK, 2024-03-28:
Locking the issue since there have bene no answers for over a week. Please reopen it if the problem still occurs for you.

closed 2024-03-28 by NoobishSVK