NoobishSVK/fm-dx-webserver #42: Default frequency at startup issue

stailus, 2024-03-25:
At startup, the tuner goes to the default frequency even if the option in the setup is not activated. The tuner is set to the default frequency (88.8 Mhz in my case), but it shows 87.5 Mhz

[image]

[image]

[image]

NoobishSVK, 2024-03-25:
I made a few changes, it should work properly now without tuning to the defaultFreq.

stailus, 2024-03-26:
Yes is ok now.

stailus, 2024-03-26:
Now if I set default freq on, webserver don't start...

[image]

NoobishSVK, 2024-03-28:
Should be fixed with v1.1.9, please check.

stailus, 2024-03-29:
Yes is ok now.

closed 2024-03-29 by stailus