NoobishSVK/fm-dx-webserver #50: Cleanup some button labels for better screen reader accessibility

marvellousmachine, 2024-05-17:
I've compiled multiple similar issues into one. I believe these are pretty easy to fix.

1. "Play / Stop Button" label still present on full screen / maximized window. Just "Play / Stop" is enough.
2. "Tune down by 100 KHz" & "Tune up by 100 KHz": Just Tune down & Tune up would be enough, I believe, As there are various steps for various bands and purposes.
3. The chat button needs a proper label. Currently it's just " (0)"or whatever number of unread messages there is. Example solution: "Chat (0 unread)".
4. The Online users button should have a number of users as part of the label. It's impossible for the screen reader user to access this information on mobile, and on PC it's also cumbersome at best.
5. The Send button in the chat area lacks the label alltogether. I recommend labelling it as "Send".

NoobishSVK, 2024-05-20:
Thanks for the feedback. Let me address all the changes. 
1. Play / Stop will be changed in the next update, as per request
2. Tune up & down buttons will also be updated 
3. The chat button will be updated, it should work properly with the new update
4. The online users button on desktop will also have a new label, but not on phone, as on the phone this is hidden to save space and this data is actually isnide the panel where you can access settings
5. Label added, will also be present in the new update

marvellousmachine, 2024-05-20:
Big thanks, everything works perfectly in 1.2.2

NoobishSVK, 2024-07-18:
Try now, i have added more info, thank you.

marvellousmachine, 2024-07-18:
Now it works properly in both views. Thanks.