NoobishSVK/fm-dx-webserver #54: Settings panel and Setup / Admin page: Checkable elements are not exposed as checkboxes to screen readers

marvellousmachine, 2024-05-17:
In both the settings panel and Admin page there are checkable elements that aren't exposed as checkboxes to screen readers. Please, consider making all such elements exposed either via ARIA: checkbox role or converting them to native HTML checkboxes.

Examples in the Settings panel: 

* ADD DECIMALS MANUALLY
* ADD UNDERSCORES TO RDS PS
* SMOOTH SIGNAL

Examples in the Admin page:

* PUBLIC TUNER (NO PASSWORD)✓
* ADMIN LOCK [ONLY ADMINS CAN TUNE]
* AUTO-SHUTDOWN

More about implementing:
https://developer.mozilla.org/en-US/docs/Web/Accessibility/ARIA/Roles/checkbox_role

NoobishSVK, 2024-07-18:
I have changed a few things related to this issue in the code, can you try to see if there was any improvement?

marvellousmachine, 2024-07-18:
Unfortunately, it remains the same.

NoobishSVK, 2024-07-31:
Please check now. I have added aria-labels to all the checkboxes

marvellousmachine, 2024-08-01:
They are present if I switch page styling in Firefox to No Style. Otherwise they are not. I know I've mentioned it on Discord, but I believe it's better to have it documented here as well...

NoobishSVK, 2025-04-22:
I have completley reworked them, I think they should work properly now. Please check.

NoobishSVK, 2025-05-30:
After talking privately, we have decided to stick with the current settings, so I'm closing it for now.

closed 2025-05-30 by NoobishSVK