NoobishSVK/fm-dx-webserver #55: Admin page tabs aren't exposed as tabs to screen readers

marvellousmachine, 2024-05-17:
In the Admin page, there are tabs such as Dashboard, Tuner, Connection etc. Please, consider making these tabs exposed to screen readers via ARIA: tab role.
https://developer.mozilla.org/en-US/docs/Web/Accessibility/ARIA/Roles/Tab_Role

marvellousmachine, 2024-05-21:
Yes, currently the tabs are properly indicated. The minor problem remains proper keyboard navigation. 

## Current situation:

* When tabbing / Shift+tabbing, every tab receives focus.
* I can activate it by pressing Enter.

## Ideal situation:

* When tabbing / Shift+tabbing, only the currently selected tab receives focus.
* I can move through all the tabs using left / right arrows
* I can activate the focused tab selected in the previous step by pressing Enter.

Thanks.

marvellousmachine, 2024-06-15:
I've checked the behavior in multiple web apps that use tabs (Azuracast, Messenger, Alza) and nobody does it right. If it's too much tinkering, feel free to leave it at its current state, it's certainly usable the way it is now.

NoobishSVK, 2024-07-18:
I have changed a few things related to this, can you try, please?

marvellousmachine, 2024-07-18:
Works great now. Thank you.

closed 2024-07-18 by marvellousmachine