NoobishSVK/fm-dx-webserver #60: Auto reconnect function

39703, 2024-06-25:
Every few days, my radio drops the COM connection with the server - the audio keeps working, but the tuning and data stops. It's always fixed by just restarting the server app, but I have to do this manually. Is it possible to incorporate code into the server that automatically attempts reconnection to the radio if it detects that the data from the COM port stops coming in? Thank you.

NoobishSVK, 2024-07-16:
I can certainly look into that, as a workaround for now you can use xdrd and the TCP/IP connection method. If it keeps crashing even with this configuration, if you're running under Windows, I recommend using WSL.

NoobishSVK, 2024-07-31:
Actually, i have checked this, sadly this issue cannot be fixed within the webserver as it's a driver issue and usually restarting the entire serial connection from the ground up fixes this. This only happens on Windows with certain devices. So I would really recommend WSL and running xdrd under that instead.

39703, 2024-07-31:
Thanks for looking into that, NoobishSVK. Unfortunately, I'm using a really old Windows desktop PC and RAM usage is already close to 100%, so I'm not sure it'd cope with WSL as well.

Would it be possible to just incorporate an option for FM-DX-WebServer to just quit if it detects that the data from the radio has stopped coming in? With that, I just need to execute the server from a looping batch file which will fix the issue also.

Thanks again.

NoobishSVK, 2024-09-26:
@AmateurAudioDude implemented something like this in the newest version, maybe this will work for you?

39703, 2024-09-28:
Thanks for the update. Yes, I've already installed it and it's been working great. Thanks again to the both of you :).

closed 2024-09-28 by 39703