NoobishSVK/fm-dx-webserver #61: Feature request: Visual presets

DABtester, 2024-07-04:
Feature request

At the moment 4 presets are possible, and can be selected with the function keys.
It would be nice if there could be 10 presets, that are visualized and selectable on the UI - with label that can be user defined

- Only number of presets that are configured will be visualized
- Mouse hover over preset will show the user defined label
- Presets can only be added/configured by admin

[image]

NoobishSVK, 2024-07-14:
Hello! This is possible to make via a plugin, maybe this is what you'e looking for? ;) It should always be possible to add new ones in the plugin config. 
https://github.com/mateuszzpl/webserver-station-memory

NoobishSVK, 2024-07-31:
i have re-checked this and it seems like it's exactly what you looking for. If you need help installing the plugin, you can ask on our Discord server at any time.

closed 2024-07-31 by NoobishSVK