NoobishSVK/fm-dx-webserver #70: Make Websocket compatible to Rdsspy

andimik, 2024-08-01:
The new Websocket has two main issues.

- RDS blocks with error in it need be indicated as `----` and not `0000`
- The syntax shall start like mentioned in the discord posting 

https://discord.com/channels/1053804249651359765/1170013490661048411/1268252572909572096

> To work with RDSSpy syntax is: `G:\r\n<BlockA><BlockB><BlockC><BlockD>\r\n\r\n`

andimik, 2024-08-02:
https://discord.com/channels/1053804249651359765/1170013490661048411/1268633081997430875

And add a reset when frequency changes.

> When changing frequency, send: `G:\r\nRESET-------\r\n\r\n`

andimik, 2024-08-02:
The first issue is solved in https://github.com/NoobishSVK/fm-dx-webserver/commit/3035013744ea5044493e027493e74de976d55c70

NoobishSVK, 2024-08-30:
With v1.2.7, the reset line was added, can you check if it works for you?

closed 2024-09-25 by NoobishSVK