NoobishSVK/fm-dx-webserver #85: The settings from the map are not applied when accessing a server

numerci, 2024-08-30:
1. Go to the servers.fmdx.org
2. Set some parameters like Signal Units = dBuV and Show RDS underscores
3.  Access some of the servers (that have 1.2.6a or 1.2.7).
Result: The preferences set in the portal have no effect on the actual servers.
Note: I would assume is a bug on the webservers side as even if these are accessed using /?signalUnits=dBuV the setting is not applied anymore.

ryanginn, 2024-09-27:
Similar issue occurs here - ISTR said features weren't included in those versions.

tomasz-tef, 2024-09-28:
opening an admin webpanel @ 1.3.1 version with #webserver already chosen always occurs with "disabled" chat label, but "enabled" chat toggle when it's original state is enabled by default @ the server

NoobishSVK, 2026-10-27:
I have closed the issue for now, as the entire feature will be completely reworked in the near future and the current one has been temporarily removed from maps.

closed 2026-10-27 by NoobishSVK