NoobishSVK/fm-dx-webserver #100: TX info data sometimes displayed on wrong frequency

AmateurAudioDude, 2024-11-07:
As previously discussed (but I'm sure you'll better appreciate it as a GitHub issue).

There's an issue with the current /server files `datahandler.js` and `tx_search.js` where sometimes TX info data is displayed on the incorrect frequency.

While tuning, if the frequency is changed either just before or just as the TX info data is displayed, it will be displayed on the changed frequency. For example, data for 93.1 might appear on 93.2.

If it's not an issue in Europe, it might possibly be more of an issue for servers located thousands of kilometres from fmdx.org, caused by high latency.

The current workaround is grabbing v1.2.8.1 files from https://github.com/NoobishSVK/fm-dx-webserver/tree/acc0c4fb165e49bc47b1d718afd625e84b53aecf/server
and using ``// Get the received TX info`` code from the older version of `datahandler.js`, and `fetchTx` & `processData` from the older `tx_search.js`.

NoobishSVK, 2025-02-09:
I plan on rewriting the entire datahandler to make it more optimized. This happens during a server lag, where it detects the TX from the data you gave it, but in the meantime you switch to a new frequency. Only happens when the data is not cached yet.

AmateurAudioDude, 2025-02-10:
That sounds correct. The lag is always there due to the latency to EU servers, meaning I'll have to stick with older code for now until there's newer versions of the files that aren't affected by it.

closed 2026-02-12 by NoobishSVK