NoobishSVK/fm-dx-webserver #107: Suppress Code 0x0000 in RDS group A

andimik, 2025-01-03:
I have realized that the Websocket transmits the Code 0x0000 in RDS group A (= PI Code), although there is no station here with this (test only) code.

```
G:
5214------------

G:
0000------------

G:
0000------------

G:
5214------------

G:
0000----86C9----

G:
52146548----2044
```

Could you suppress this or replace by `----` because this leads to wrong logs? See screenshot from RDSSpy.

[image]

[rdssocket.txt](https://github.com/user-attachments/files/18299648/rdssocket.txt)

ryanginn, 2025-01-13:
Maybe something to be looked into @mrwish7

andimik, 2025-03-21:
Due to the fact that Sjef's Firmware 2.20 now supports 0x0000 (we already have some confirmed PI Codes, checked with Rdsspy as well), I just wanted to ask you for an update on this.

NoobishSVK, 2025-04-19:
I have asked @mrwish7 to check this again, once it's done, I will merge it immediately without a version bump.

mrwish7, 2025-04-21:
If I'm understanding it correctly, I think that this group A 0x0000 bug could potentially be fixed just by modifying line 355 in server/datahandler.js -

```
} else {
   pi = '0000';
   errorsNew = (0x03 << 6);
}
```
The `pi = '0000'` can just become `pi = '----'` instead. I'm not sure exactly how the logic triggers in this but it seems in theory that if there was a valid PI 0000 it would trigger the 'if' and work correctly (I guess that else just puts 0000 when a PI can't be decoded properly?). If so, both elements of the bug would be fixed.

I didn't write any of this though so I'm not sure if it affects anything else? The code in question all seems to be inside the handleData function.

closed 2025-04-22 by NoobishSVK