NoobishSVK/fm-dx-webserver #110: Bandwidth switcher broken?

cazian, 2025-01-12:
The bandwidth switcher appears to be broken in version 1.3.3. I highlight the option in the settings and click the save icon, but the bandwidth switch no longer appears on refreshing the webserver page or even after rebooting the server. If the config page is refreshed after the save, the bandwidth switcher button is unhighlighted again.

ryanginn, 2025-01-13:
Can second this, does not work for me either !

ryanginn, 2025-01-13:
That is one way to fix it, but the main question is **what** is causing bwSwitch to not equal true? I wonder is it possible to force it to true in the config file? Never tried.

Ryan

Overland-DX, 2025-01-13:
I don't know. I had the exact same problem in V1.3.2 too. It seems that either "tef" or "bwSwitch" is enabled. It can't detect if it is enabled.

ryanginn, 2025-01-13:
Interesting

ryanginn, 2025-01-13:
OK. that is interesting. Maybe this will help @NoobishSVK find the problem

Overland-DX, 2025-01-13:
I hope so. I have enabled bwSwitch, but it is not saved in the config file.

Is it in your config file, @ryanginn

cazian, 2025-01-13:
I can confirm the bandwidth switch is busted for me as well in 1.3.3.
Thanks for the workaround tip I'll try this one I get home.

Best regards
Ian Kelly

NoobishSVK, 2025-01-13:
I will fix it during this week, thank you for notifying me.

ryanginn, 2025-01-13:
Not showing in mine either

Le meas,
Rhían Mac Guinneál

ryanginn, 2025-01-13:
> I hope so. I have enabled bwSwitch in adminpanel, but it is not saved in the config file.
> 
> Is it in your config file, @ryanginn ?
> 
> I think it should actually be in this section at the bottom of the file. ![image](https://private-user-images.githubusercontent.com/102610897/402500284-dae2c302-473f-4c98-b177-1b79e6ca46bc.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3NzQ0MzEsIm5iZiI6MTczNjc3NDEzMSwicGF0aCI6Ii8xMDI2MTA4OTcvNDAyNTAwMjg0LWRhZTJjMzAyLTQ3M2YtNGM5OC1iMTc3LTFiNzllNmNhNDZiYy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwMTEzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDExM1QxMzE1MzFaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xMWVjYWRiNGI1ZDU2NjQ0YTgxMDJhZjhkNTNhNTI3MGNkZDMwMjk1NzYzZDg4MjdiMGI2OWMyMjQwNDM5MDE2JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.nzYa7eeVkk8omjEi7XAMLhco_AHLJQZ6O5vfkfcYs0A)

Not showing in mine either when checked

cazian, 2025-01-14:
I've only just had a chance to try the two workarounds suggested above. For some reason, the first one suggested by @Overland-DX didn't bring the bandwidth switch to life but the second one did the trick for me in getting it working again. Thank you. I do hope that @NoobishSVK can solve the problem so the workaround isn't needed, good luck!

NoobishSVK, 2025-01-16:
Please check now, it should work via admin settings in 1.3.3.1

NoobishSVK, 2025-01-16:
Closing as per multiple reports on Discord of it being fixed with the update.

closed 2025-01-16 by NoobishSVK