NoobishSVK/fm-dx-webserver #121: Preset frequencies get lost

BVE-FM, 2025-01-21:
When set 1 till 4 preset frequencies and save the config they are working fine.
When log out of the config and log back in ... all presets are gone.

NoobishSVK, 2025-02-09:
Is this is still an issue in v1.3.3.1? I believe I fixed this.

BVE-FM, 2025-02-09:
> Is this is still an issue in v1.3.3.1? I believe I fixed this.

Yes, it’s still an issue

NoobishSVK, 2025-02-09:
Can you try to update to the latest hotfix version? I made some hotfixes after 1.3.3.1 and I believe this was one of them.

BVE-FM, 2025-02-09:
> Can you try to update to the latest hotfix version? I made some hotfixes after 1.3.3.1 and I believe this was one of them.

I”ll try in Tuesday. Right now I’m in The Gambia, Africa.

NoobishSVK, 2025-02-09:
Would be awesome to have a webserver there :) 
Anyway, I am closing this as multiple people have reported it's been fixed.

closed 2025-02-09 by NoobishSVK