NoobishSVK/fm-dx-webserver #122: Running multiple public servers, different port, same IP

BVE-FM, 2025-01-21:
When running multiple servers from same IP, same PC but with different port numbers only the last started server will be visible in the public server list.

NoobishSVK, 2025-01-22:
Is it possible that you are using the same token for both servers for the map? Maybe that's why, if you copied the config from the first server.

BVE-FM, 2025-01-22:
> Is it possible that you are using the same token for both servers for the map? Maybe that's why, if you copied the config from the first server.

Indeed, after I got the first server running I copied the files to a different folder, changed the config (port nr.) and run. Is there an easy way to solve this issue or do I need to do a full new install?

NoobishSVK, 2025-01-24:
You can just run 2 different configs with the --config parameter on startup. For example "node . --debug --config server2"
This will generate a new config file. I recommend going through the wizard rather than copying the old config in this case.

closed 2025-01-24 by BVE-FM