NoobishSVK/fm-dx-webserver #125: Backwards tuning with 'B' hotkey not working in conjunction with AF clicking

finndx, 2025-02-17:
This issue has been already reported. 
If you want to check for AF frequencies fast and efficiently, using the hotkey 'B' for tuning backwards isn't working the way it is assumed to work.

Steps:

1) You have tuned to the frequency whose AF frequencies you want to check.
2) You click any of the frequencies in the decoded AF list => frequency changes correctly to the chosen AF frequency. You'll either hear the same program on this frequency or not.
3) You want to get that AF list back for checking whether another AF frequency is audible. So you type 'B' for the previous frequency.

At this point, it tunes to the frequency one step before your frequency whose AF you really want to check for. So practically, you lost the frequency whose alternative frequencies you are checking, because it tunes one extra step backwards.

This means that clicking the AF frequency doesn't register the tuning correctly/properly.

Should be easy to fix.

NoobishSVK, 2025-04-19:
fixed with the new webserver version

NoobishSVK, 2025-05-04:
As reported privately by @finndx  - now it doesn't react during different circumstances, such as entering the freq manually via input.

NoobishSVK, 2025-05-30:
Fixed now.

closed 2025-11-09 by NoobishSVK