NoobishSVK/fm-dx-webserver #139: Audio dropouts

m0euk, 2025-04-21:
Mainly occuring with fairly strong signals, generally not weak ones or no signal, any device or browser seems the same. See my server io95.fmtuner.otg.

https://github.com/user-attachments/assets/71587366-aaf5-41f3-8536-3b32741e34da

. See video for example. I'm running on an RPi if that's of any use. Thanks.

m0euk, 2025-04-22:
Update, this seems to be a function of the volume level on the silver TEF, see video, where vol is turned from low to high, Setting at a mid level fine, too hight and it drops out. So a function of audio overload when a signal is strong rather than an issue/bug it seems. Thx.

https://github.com/user-attachments/assets/1e868738-28d2-4fe9-a1e3-36f7ecdb623e

NoobishSVK, 2025-04-27:
This could be something that happens before the webserver gets the audio, either in FFMPEG, or direct audio capture. Do you perhaps have a machine where you could try to capture the audio separately directly from the TEF?

m0euk, 2025-04-27:
I have a Mac or Windows machine I can use if you tell me what you need me to do (or the RPi running the server). Thanks.

m0euk, 2025-05-01:
> Before we continue, do you perhaps have the exact hardware your webserver is running on? I suspect there might be some kind of a CPU / RAM limitation with more dynamic passages, which causes audio dropouts, as the webserver seems to load really slow as well.

Yes it is a Raspberry Pi 2B, 1Gb RAM, running latest version of the server and all plugins up to date. Note that now I’ve moved to a headless TEF and seen no further issues related to this.Thanks. (graeme_ncl on Discord / io95.fmtuner.org)Sent from my iPhone.

NoobishSVK, 2025-05-01:
In that case it might have been something hardwsare related rather than webserver, since you aren't seeing this issue after upgrading your hardware anymore, I am closing this, thank you.

closed 2025-05-01 by NoobishSVK